LELEC210X/LELEC210X #12: Error when running this command for poetry:poetry run python -m ipykernel install --user --name LELEC210X

nonoswagman, 2023-11-25:
After installing poetry (I use visual studio code bash terminal on Windows), This  error  appairs when I run the command :run python -m ipykernel install --user --name LELEC210X
ImportError: DLL load failed while importing _sqlite3: Le module spécifié est introuvable.
I tried to solve it but without succes.

jeertmans, 2023-11-25:
The error is caused by a Python installation that did not come with sqlite3 support, which is needed for some dependencies.

The best solution is to install Python with sqlite3 support, but this should not be needed on Windows I think (only on Linux).

can you check this solution ? https://stackoverflow.com/questions/54876404/unable-to-import-sqlite3-using-anaconda-python?answertab=active#tab-top

jeertmans, 2023-11-30:
@nonoswagman did you solve your issue?

closed 2024-02-05 by jeertmans