LELEC210X/LELEC210X #27: PortAudio not found

Rauwla, 2024-09-21:
Hello, 

While trying to run h1_feature_vectors with rye, the following error appears (see the image).
[image]

Can you help me? 
Thanks

jeertmans, 2024-09-21:
Hello @Rauwla, it means that you have to install PortAudio libraries, see https://github.com/Upsonic/gpt-computer-assistant/issues/22#issuecomment-2147378454.

What OS / platform are you using? WSL / Linux / macOS ?

Rauwla, 2024-09-22:
Thanks I followed the instructions and now it works!

jeertmans, 2024-09-22:
Happy it works! Can you just let me know what OS / platform you are using?

Rauwla, 2024-09-28:
Of course, I'm using Linux.

jeertmans, 2024-09-28:
Native Linux (like a dualboot), or Linux in a Virtual Box / inside WSL?

Rauwla, 2024-09-28:
Yes it's a dualboot so native linux

closed 2024-09-28 by jeertmans