LELEC210X/LELEC210X #30: Installing GNU Radio 3.8 on Linux versions > 20.04

MClotuche, 2024-09-29:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue+label%3Adocumentation+) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

I attempted to install GNU Radio 3.8 on my Linux Ubuntu 22.04.5 dual-boot system, but no matter what method I try, GNU Radio 3.10 gets installed instead. This is problematic since the project specifically requires version 3.8.

The standard installation method is using the 'apt' package manager. However, on my Ubuntu version (22.04.5), which is newer than 20.04, GNU Radio 3.8 is not available, so an older version gets installed. This behavior is expected.

After uninstalling the wrong version of GNU Radio, I tried installing the software from source, as recommended on the Moodle forum and explained on the GNU Radio wiki: [GNU Radio Wiki - Linux Install for Version 3.8 or Earlier](https://wiki.gnuradio.org/index.php?title=LinuxInstall#For_GNU_Radio_3.8_or_Earlier). I followed each step in the guide, starting with cloning the GitHub repository, checking out the 3.8 version, running make -j3, sudo make install, and setting the Python path. I encountered some issues with setting the Python path, but I resolved them by creating a symbolic link to the software in the usr/bin directory, which solved that problem.

However, when I launch the application, GNU Radio 10.10.1.1 opens instead of version 3.8. I double-checked that the version installed via 'apt' was properly uninstalled by repeating the process twice. Based on this, I suspect that even when installing from source, version 10.10.1 is somehow being installed, which seems unusual.

If any other student or teaching assistant has successfully installed version 3.8 on a more recent Linux distribution than 20.04, I would greatly appreciate it if you could share your method!

### Command or program

GNU Radio 3.10

### Issue Type

Installation issue

### What is your platform?

Linux (dual boot)

### Screenshots

[image]

jeertmans, 2024-09-30:
Hello @MClotuche, thanks for reporting that issue!

I did some tests locally (Ubuntu 24.04) and I cannot build from `origin/maint-3.8`, apparently because they use old C code that does not compile anymore with GCC 13. It might be possible to manually downgrade GCC, I haven't tested this yet.

What is the output of `gcc --version`?

Did you actually successfully compile? Because this doesn't make sense that you have GNU Radio 3.10 installed, if you were really from the `maint-3.8` branch.

Did you check that you were indeed on the correct branch?

We are currently stuck with GNU Radio 3.8 because LimeSDR doesn't work with any earlier version, but on the other hand GNU Radio 3.8 starts to be too old to be maintained :-(

jeertmans, 2024-09-30:
> [!NOTE]
>
> It might be worth testing the PyBOMBS installation: https://github.com/gnuradio/gnuradio/tree/maint-3.8?tab=readme-ov-file#pybombs.
>
> I tried to apply the necessary patch to have GCC 13 compatible on the maintainance branch,
> you can try and see if that works by cloning my fork:
> https://github.com/jeertmans/gnuradio/tree/maint-3.8

jeertmans, 2024-10-22:
With #42, @PolMaistriaux documented a solution to run the project with GNU Radio 3.10, which is available on Ubuntu > 20.04.

I will close this now, as there exist a workaround, and it might be too much work to try to port GNU Radio 3.8.

closed 2024-10-22 by jeertmans