LELEC210X/LELEC210X #33: Errors of includes while building the hands_on_audio_acquisition

ppasture, 2024-10-08:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue+label%3Adocumentation+) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

Hello,

We tried to build the hands_on_audio_acquisition for the H2b part of the project but we get some issues with including the "stm32l4xx_hal.h" and "stm32l4xx.h" with "no such file or directory". We have already installed Software Package stm32l4 for the hands_on_mcu from last week.

Thank you!

### Command or program

build the hands_on_audio_acquisition

### Issue Type

Error when running an command

### What is your platform?

Linux (WSL)

ppasture, 2024-10-08:
Hi @jeertmans,

Sorry for the inaccuracy, my STM IDE is well installed on Windows OS. The project is updated and it works for 2 of our 4 computers. We are all on Windows and we all installed the package stm32l4..

jeertmans, 2024-10-08:
Can you edit the main post and include one or two screenshots showing what you do and the error message in its context?

You can easily copy-and-paste or drag-and-drop images.

ppasture, 2024-10-09:
I inserted a screenshot of the console @jeertmans !
Thank you :))

brockauly, 2024-10-09:
Hi @ppasture ,
If the steps given by @TheTarados do not solve your problem, you can try these ones.
Did you clone the git project on your Linux volume through WSL ? If yes, then this can be the cause. Windows has some problems to access Linux file, but not the other way around. That's why you should install as many softwares as possible on your host OS, including git and a git GUI.

ppasture, 2024-10-09:
Hi @TheTarados @brockauly,

Problem fixed, I had not understood how the .ioc file was working. I generated the code with the .ioc as you told me to do and now it works!

Thank you,
Pierre

jeertmans, 2024-10-09:
Perfect, can you include a screenshot of how you did generate the code with the `.ioc`? So I can mark this as "solved" and close the issue :-)

ppasture, 2024-10-09:
After opening the .ioc file (and migrate the project if it has to), I generated the code using the "Project->Generate code" button on the top tab.

Problem solved now!

[image]

closed 2024-10-09 by jeertmans