LELEC210X/LELEC210X #39: Wrong installation path for out-of-tree modules (FSK) - GNU Radio

Reshims, 2024-10-21:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue+label%3Adocumentation+) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

During CMake installation, the python library path (the likes of `/usr/local/lib/pythonx`, where `x` is some version number) which is used to install out-of-tree modules (FSK) for GNU Radio is different than the one used by python.

In my case, it defaults to `/usr/local/lib/python3` even though the library path used by my python versions (`/usr/bin/python3` **and** `/usr/bin/python3.8`) is `/usr/local/lib/python3.8`.

### Command or program

cmake ..

### Issue Type

Installation issue

### What is your platform?

Linux (WSL)

### Relevant error output

```shell
GNU Radio `fsk` blocks (e.g. 'Demodulation') missing.

Simalarly, when launching a GNU Radio python script directly from the terminal, the following error occurs:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'fsk'

when importing the module `fsk` in python.
```

### Screenshots

Missing blocks in GNU Radio script "decode_capture.grc"
[image]

### Additional information

Changing manually the python executable in the CMakeLists.txt (`set(PYTHON_EXECUTABLE "/usr/bin/python3.8")`) before or after defining the python executable for GNU Radio (`find_package(Gnuradio "3.8" REQUIRED)`) doesn't seem to work.

PolMaistriaux, 2024-10-21:
Hi,

The pdf for the Hands-On H3b should be updated today with various information and also a fix for this issue.

Thanks for letting us now,

Pol

jeertmans, 2024-10-21:
@Reshims do you have multiple Python installations? Usually, their binaries located in `/usr/bin`.

What you do is indeed not the correct solution (it works, but meh). If I remember correctly, there is an option inside GNU Radio that lets you customize the path to your Python installation. You should check this out.

Reshims, 2024-10-21:
After checking, I have only one, which is python 3.8. My python 3 version is just a symbolic link (I think... Looks like it but I'm no expert). 
[image]

So it makes sense that changing the python executable directly in the CMakeLists didn't do anything...

jeertmans, 2024-10-21:
You need to setup the absolute path to the `third_party` folder, so `sudo make install` actually installs the library there. This is documented [in the Wiki](https://github.com/LELEC210X/LELEC210X/wiki/Wireless-Communications#gnu-radio----no-module-named-fsk) actually :-)

Reshims, 2024-10-21:
I missed that info, my bad. Thanks for the help !

jeertmans, 2024-10-22:
Nice! @PolMaistriaux included this information in the H3b, now on Moodle.

Closing this issue as it is now solved.

closed 2024-10-22 by jeertmans