LELEC210X/LELEC210X #47: ERROR - STM32: no file stm32l4xx_hal.h found when compiling

MClotuche, 2024-10-24:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue+label%3Adocumentation+) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

When trying to build any project file (ex. hands_on_main_app), the console gives me several errors : 
``../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory.`

I searched in all my LELEC210X files, and indeed, the required header file doesn't exist. 

Analysis and what I've tried:

1. I already had the same problem two weeks ago (when I was on dual-boot linux, now I am on windows so the OS isn't the problem), and I solved it by copying the ___stm32l4xx_hal_ file that existed in another project (ex: it worked with hands_on_mcu so the file was created in this project). However, this time the file doesn't exist at all in my LELEC210X folder (I used search methods).

2. Two other members of my group had the same problem. For one, it was solved by regenerating the code of the ioc file. However, it didn't work for two of us. 

3. I tried to erase the project, connect the mcu board, recreate the project, clean it, regenerate the code with the .ioc file, the build the project, but it didn't work.

### Command or program

make -j8 all 

### Issue Type

Error when running an command

### What is your platform?

Windows

### Relevant error output

```shell
arm-none-eabi-gcc "../Core/Src/adc.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/adc.d" -MT"Core/Src/adc.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/adc.o"
arm-none-eabi-gcc "../Core/Src/adc_dblbuf.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/adc_dblbuf.d" -MT"Core/Src/adc_dblbuf.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/adc_dblbuf.o"
arm-none-eabi-gcc "../Core/Src/arm_absmax_q15.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/arm_absmax_q15.d" -MT"Core/Src/arm_absmax_q15.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/arm_absmax_q15.o"
arm-none-eabi-gcc "../Core/Src/dma.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/dma.d" -MT"Core/Src/dma.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/dma.o"
arm-none-eabi-gcc "../Core/Src/eval_radio.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/eval_radio.d" -MT"Core/Src/eval_radio.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/eval_radio.o"
arm-none-eabi-gcc "../Core/Src/gpio.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/gpio.d" -MT"Core/Src/gpio.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/gpio.o"
arm-none-eabi-gcc "../Core/Src/main.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/main.d" -MT"Core/Src/main.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/main.o"
arm-none-eabi-gcc "../Core/Src/packet.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32L4A6xx -c -I../Drivers/CMSIS/Include -I../Core/Inc -I../Drivers/CMSIS/Device/ST/STM32L4xx/Include -I../Drivers/STM32L4xx_HAL_Driver/Inc -I../Drivers/STM32L4xx_HAL_Driver/Inc/Legacy -I"C:/Users/marti/OneDrive/Documents/Ecole/EPL/M1/Q1/LELEC2102-Project_Embedded_system/LELEC210X/mcu/hands_on_main_app/ARM_CMSIS/CMSIS/DSP/Include" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/packet.d" -MT"Core/Src/packet.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/packet.o"
In file included from ../Core/Inc/dma.h:28,
                 from ../Core/Src/dma.c:21:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
In file included from ../Core/Inc/adc_dblbuf.h:4,
                 from ../Core/Src/eval_radio.c:5:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from ../Core/Inc/adc.h:28,
                 from ../Core/Src/adc.c:21:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
In file included from ../Core/Src/main.c:21:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from ../Core/Inc/gpio.h:28,
                 from ../Core/Src/gpio.c:21:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
[... 7 lines cut ...]
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from ../Core/Src/arm_absmax_q15.c:30:
[... cut ...]
make: *** [Core/Src/subdir.mk:76: Core/Src/arm_absmax_q15.o] Error 1
make: *** [Core/Src/subdir.mk:76: Core/Src/adc_dblbuf.o] Error 1
In file included from ../Core/Src/packet.c:8:
../Core/Inc/main.h:31:10: fatal error: stm32l4xx_hal.h: No such file or directory
   31 | #include "stm32l4xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Core/Src/subdir.mk:76: Core/Src/packet.o] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

13:37:47 Build Failed. 17 errors, 0 warnings. (took 780ms)
```

PolMaistriaux, 2024-10-24:
Hi, 

I suspect you do not have the required drivers in the project which should normally be fixed by regenerating the code, just to be sure :

[image]

Do you have the following structure in STM32 Cube IDE, with the "Drivers" folder missing? If present (but you have the compilation error you mention), right click on it and delete it. Also, go to this folder via the file explorer and delete the "Drivers" folder if it is present there.

Then regenerate the code by : 

- Double-clicking on the .ioc
- On the top bar, between "Search" and "Run", do "Project > Generate Code".

[image]

Do you still have the error now?

TheTarados, 2024-10-24:
Duplicate of #33

MClotuche, 2024-11-12:
Thank you for your answers.
 
The problem was that some updates of the STM32 software were not done, so generating the code did not produce any file at all, which caused the error. 

After installing the updates (it took some time), "generate code" indeed created the recquired file for compiling.

closed 2024-11-12 by TheTarados