LELEC210X/LELEC210X #48: Matplotlib unable to plt.show()

JeromeHen-UCL, 2024-10-31:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

When trying to show a plot via `plt.show()`, our group gets the error 

> /home/jehendrick/Documents/EPL/Q7/LELEC2910/Results/plotter.py:37: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown

This issue is present on all of our systems. For now we have to `plt.savefig("...")` and manually open the resulting figure. This issue wasn't present before installing Rye.

### Command or program

plt.show()

### Issue Type

Error when running an command

### What is your platform?

Other (please precise below)

### Other platform

Windows+WSL or native Ubuntu 20.04 or Debian + Ubuntu 20.04 (Virtualbox)

### Relevant error output

```shell
/home/jehendrick/Documents/EPL/Q7/LELEC2910/Results/plotter.py:37: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown
  plt.show()
```

### Additional information

We tried to change the backend used by matplotlib to "QtAgg", but we were unsuccessful or had new errors stating missing packages.

jeertmans, 2024-10-31:
Hello @JeromeHen-UCL, are you sure you are running code from our project? Because

```
/home/jehendrick/Documents/EPL/Q7/LELEC2910/Results/plotter.py:37:
```

does not look like LELEC210X.

If you Google your solution, you find plenty of solutions, like see the comments here: https://stackoverflow.com/q/77507580.

If this issue is related to this project, please provide the command and files you executed, otherwise it's hard to help.

JeromeHen-UCL, 2024-11-01:
Hi @jeertmans, sorry I forgot to mention that the issue was system-wide instead of just in LELEC210X.

I used to run `rye run python sim.py` (from LELEC210X/telecom/hands_on_simulation) and I used to have these errors:

> /home/jehendrick/Documents/EPL/Q7/LELEC2102/LELEC210X/telecom/hands_on_simulation/sim.py:282: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown
>   plt.show()
> /home/jehendrick/Documents/EPL/Q7/LELEC2102/LELEC210X/telecom/hands_on_simulation/sim.py:361: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown
>   plt.show()
> /home/jehendrick/Documents/EPL/Q7/LELEC2102/LELEC210X/telecom/hands_on_simulation/sim.py:371: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown
>   plt.show()
> /home/jehendrick/Documents/EPL/Q7/LELEC2102/LELEC210X/telecom/hands_on_simulation/sim.py:381: UserWarning: FigureCanvasAgg is non-interactive, and thus cannot be shown
>   plt.show()

Thanks for the solution link! I somehow did not read that the newer package was PyQt6 instead of PyQt5 and this made a mess when I tried to install the older version.

I ran `rye install PyQt6` and now it works as expected (run `pip install PyQt6` for "classical" venv).

closed 2024-11-01 by JeromeHen-UCL