LELEC210X/LELEC210X #49: Error when compiling the project on quartus

lraviart, 2024-11-08:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

When trying to compile the design on Quartus, errors appear and I am unable to compile it. This is also the case for all the members of our group. We think we did all the steps described in the H4 assignment.

### Command or program

Quartus

### Issue Type

Error when running an command

### What is your platform?

Windows

### Screenshots

[image]
[image]

### Additional information

We tried it with the design from our forked github and from the original one so we don't think it is an update problem.

lraviart, 2024-11-13:
We found the problem, when opening the project Quartus offers to do an IP upgrade which causes the error. The fix is simply not to do it. 
[image]

jeertmans, 2024-11-14:
It indeed reminds me of some problem we encountered in the past. Glad you found the solution :)

closed 2024-11-14 by jeertmans