LELEC210X/LELEC210X #50: Empty mel-spectrogram for H5a.2 part 2.1

Rauwla, 2024-11-20:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

I'm running the H5a2 file with rye on wsl, and even thought we completed the functions in the audio_student.py file, the mel-spectrogram is completeley empty, as shown on the picture. After printing the content of the dataset, we see that the inside of the classes of the dataset are empty (with the command print (dataset['fire'][0])  for example)
My teammate is also running the jupyter notebook with rye but wit windows, and he has the same problem. Could you help us? 
Thanks

### Command or program

h5a2_audio.ipynb

### Issue Type

Unexpected results (e.g., packets are wrong)

### What is your platform?

Linux (WSL), Windows

### Screenshots

[image]

jeertmans, 2024-11-21:
What does this cell output?

```python
### TO RUN
dataset = Dataset()
classnames = dataset.list_classes()

print("\n".join(classnames))
```

You confirm you are running https://github.com/LELEC210X/LELEC210X/blob/main/classification/h5a2_audio.ipynb?

Rauwla, 2024-11-23:
Yes I was, but I finally debugged it, there was a problem with the functions we wrote. 
Thanks!

closed 2024-11-23 by Rauwla