LELEC210X/LELEC210X #52: Problem when launch gnu radio with the updated file synchronization.py

lver16, 2024-12-06:
### Terms

- [X] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [X] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [X] Updated my Python virtual environment with `rye sync`;

### Describe the issue

Hello,

I have an issue when I launch the file eval_limesdr_fpga.grc : 
Traceback (most recent call last):
  File "/mnt/c/Users/lucas/Documents/LUCAS/MASTER_1/Projet/LELEC210X_Groupe_A/telecom/hands_on_measurements/gr-fsk/apps/eval_limesdr.py", line 337, in <module>
    main()
  File "/mnt/c/Users/lucas/Documents/LUCAS/MASTER_1/Projet/LELEC210X_Groupe_A/telecom/hands_on_measurements/gr-fsk/apps/eval_limesdr.py", line 315, in main
    tb = top_block_cls()
  File "/mnt/c/Users/lucas/Documents/LUCAS/MASTER_1/Projet/LELEC210X_Groupe_A/telecom/hands_on_measurements/gr-fsk/apps/eval_limesdr.py", line 155, in __init__
    self.fsk_synchronization_0 = fsk.synchronization(data_rate, fdev, samp_rate, hdr_len, packet_len, tx_power)
TypeError: __init__() missing 1 required positional argument: 'enable_log'
libusb: warning [libusb_exit] application left some devices open

>>> Done (return code 1)

I saw that in the synchronization.py, there is a new argument 'enable_log', I changed the file but I have this error. 
Do you know what can I do ?

Thank you in advance,

Lucas Verheyen, groupe A

### Command or program

launching gnu radio

### Issue Type

Error when running an command

### What is your platform?

Windows

jeertmans, 2024-12-06:
Anyway, it looks like there is a mismatch between your class definition and the instantiation.

Maybe you need to recompile / reinstall the module? Because, according to the last commit https://github.com/LELEC210X/LELEC210X/commit/d8be515eb7198595f4a3094a7a0a34f4d2dd5b42#diff-62d4d8a7c7126a4c0d03233363c397ad8e65cca02143ca28a9b8b02d665193c3, @PolMaistriaux did include the new parameter in the function signature. The Python files in `apps/` are automatically generated, so it is probably better to just re-generated them.

Let me know if that works.

lver16, 2024-12-07:
> Vous n’obtenez pas souvent d’e-mail à partir de ***@***.*** Pourquoi c’est important<https://aka.ms/LearnAboutSenderIdentification>

It doesn't work. I have a problem with the board. When I plug the Nucleo into my computer, it blinks red, but when I connect it to another computer, it connects normally.

Maybe you need to recompile / reinstall the module? Because, according to the last commit d8be515<https://github.com/LELEC210X/LELEC210X/commit/d8be515eb7198595f4a3094a7a0a34f4d2dd5b42>, @PolMaistriaux<https://github.com/PolMaistriaux> did include the new parameter in the function signature. The Python files in apps/ are automatically generated, so it is probably better to just re-generated them.

PolMaistriaux, 2024-12-09:
I am not sure to understand the issue at the moment. 
If this is still a problem at run-time of the flowgraph, as you mentioned in your first post, have you tried to recompile and regenerate the flowgraph before running it? Have you pulled the last commit?
If this is a problem with the board, this is more concerning.
In all cases, if the issue has not been solved, you can send me an email so that we can fix an appointment to look at the issue.