LELEC210X/LELEC210X #59: Uart-reader printing in white text over a white background

carbonnelleg, 2025-03-10:
### Terms

- [x] Checked the [existing issues](https://github.com/LELEC210X/LELEC210X/issues?q=is%3Aissue) and [discussions](https://github.com/LELEC210X/LELEC210X/discussions) to see if my issue had not already been reported;
- [x] Read the [best practice](https://github.com/LELEC210X/LELEC210X/wiki/Best-practice-guide) and checked if my bug was not one of the common bugs documented in the Wiki;
- [x] Updated my Python virtual environment with `rye sync`;

### Describe the issue

I am using the latest, incredibly performing uart-reader version, everything is working fine, but the console prints are in white and I have a white background. Would it be possible to update it ?

### Command or program

rye run uart-reader

### Issue Type

Unexpected results (e.g., packets are wrong)

### What is your platform?

Windows

### Relevant error output

```shell

```

### Screenshots

[image]

jeertmans, 2025-03-10:
This is probably an issue related to the current theme (i.e., light vs dark theme) you are using. Usually, Qt handles both dark and light themes pretty well by default, but hard-coding colors overrides this.

Pinging @vctrcc, the author of this tool.

vctrcc, 2025-03-10:
This bug is due to a fix for darkmode i did, ill create a pull request and fix 2 other problems while i'm at it.

closed 2025-03-11 by jeertmans